twlostow/dsi-shield #3: License - CERN OHL or GPL?

evancox10, 2016-10-21:
Hi, I have a question about which license applies to the design files and documentation. The PCB has a silk screen that states its licensed under CERN OHL v1.2. However, the LICENSE file at the top of the repo is GNU LGPL v3. Can you clarify what the license is for the material?

Thanks,
Evan

evancox10, 2016-10-21:
Woops. If I had read the first page readme I would have seen that you covered this. Sorry!

Also, I noticed that many of the files in the IP Cores folder have no license information, some have a LGPL v2.1 license. I wanted to double check that incorporating these into this LGPL v3 project was ok. According to https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility , does the combined work need to be released as GPLv3? Am I misunderstanding this?

Second, do you have any good resources you could point me to on the use of LGPL licensed HDL in an ASIC? This is clearly an evolving legal issue, and there may be no definitive answers. I am primarily concerned with the usage of the HDL in a commercial ASIC and whether or not that virally infects the rest of the chip.

Third, why not release the HDL under the CERN OHL? Do you know what CERN's position is on using the OHL vs (L)GPL for HDL?

From what I can tell, you seem to be at the forefront of many of these issues :)

closed 2017-03-31 by evancox10

dreamcat4, 2019-04-13:
Hi there. Stumbled across this project today while looking for a way to drive smartphone displays. And it seems like the only one!

In regards to licensing, it would be nice to take a general interest in these licensing and similar issues. And just remain contactable into the future about it. Which will help these project(s) to continue moving forwards without you, for other people to contribute their own work and improve the solution further.

Because there is a difference between Copyright and Licensing. The above is a good example of the linking clause being an issue. Well what to do about it?

https://en.wikipedia.org/wiki/GPL_linking_exception

This requires amendment(s) to the existing license, which can only be performed by the original (legal) copyright holder. So remaining contactable is important. Otherwise who else will want to come along be motivated to make a v3.0 with improved performance? Maybe on a bigger / faster FPGA ?

Because it seems one of the more significant use cases for this device would be 'open source VR heatsets'. However that really requires some extra performance above what was comented in the project's log over on the hackaday pages.

dreamcat4, 2019-04-13:
@twlostow In fact, this project might make for a good fit for the RISC-V open hardware project

https://en.wikipedia.org/wiki/RISC-V

So please also consider talking with stakeholders and experts over there, to see if there is any interest to develop this further. If you are looking for a more reputable place for this work to find a new home. Kind Regards