twlostow/dsi-shield #6: Adding Galaxy S4 AMOLED Adapter

Marccos, 2017-07-29:
Hi twlostow,

I saw at hackaday, that you managed to get a S4 Galaxy display working. 
https://hackaday.io/project/364-mipi-dsi-display-shieldhdmi-adapter/log/16504-galaxy-s4-amoled-screen-working

I am interested to use this display for a small project. Could you upload the schematic/layout for this adapter or give me some hints where to find datasheets for this screen? if you give me some hints where to start, I can redesign the adapter(in KiCad) and upload my results in this case.

What changes in hdl/software need to be made to use this display (with revision 1.1 which is uploaded)?

--Marccos

BiatuAutMiahn, 2017-07-29:
I too am interested in this

BiatuAutMiahn, 2017-11-19:
Any updates on this?

mybays, 2019-08-31:
I too am interested in this