twlostow/dsi-shield #8: unrecognized command line option '-m32' and '-msoft-float'

archerby, 2017-11-03:
How did you configure the toolchain? If act on the guide from the riscv site, then there is no support for the flags -m32 and -msoft-float.

twlostow, 2017-11-03:
I think the makefile uses an old RISC-V toolchain, the newer GCC has different options. Look in the GCC manual.

archerby, 2017-11-13:
The problem was in the new version of toolchain. It was enough to build a 2015 version of the toolchain and it all worked. There is an instruction that came to me here: https://github.com/dornerworks/seL4-riscv-mk/blob/july_2015_riscv_rocket/README.md

argoslot, 2018-10-18:
Hi,

I read the guide here: https://hackaday.io/project/364/instructions, but don't quite understand what to do with the buildchains.  I've been compiling Linux kernels, other blather for years -- but this is my first foray into the embedded / spartan world.

archerby, I looked at the link you provided -- and don't get how the above toolchain fits in.

I see only these two lines for how to setup the toolchain:
- Install an LM32 toolchain. You may find one on Lattice website. There's a pre-built version for Linux (IA32) available here.
- Download and install Xilinx ISE 14.7 Webpack from Xilinx website.

The lm32 toolchain above has no info/docs with it.  The Xilinx toolchain?  Well, it seems to be more of a design/gui tool.

It's clear that you guys have been working with this stuff for a long time, but there's zero info for someone to build and flash this, unless they move on to another well documented project first -- and then come back here after at least figuring things out.

If you guys could kick me a few pointers... I'd definitely to document the whole build process under Linux.  We could then add this to the repo at least.... and it'd be around for others.

Heck.. even if you can point me to guides using the above buildchains for a fairly similar project, then I can at least get some traction, but at this point I'm not even sure what missing bits are.. missing.

To be honest, as I've never worked with Xilinx/Spartan stuff.. I don't even know what flashing hardware is required here.  I'm sure you guys can just go to a shelf and find one, but I want to make sure I don't spend cash on the wrong tool.

Is it this? https://www.xilinx.com/products/boards-and-kits/hw-usb-ii-g.html

Any help is appreciated -- and, any help you give I will certainly pour back into documentation for this project.

Thanks