twlostow/dsi-shield #13: Discord?

TheAnimatrix, 2020-09-14:
This project is exactly what i've been searching for, i'm building my own graphics controller for a 7" TFT i ordered from alibaba. I'd love to interact and ask a few questions regarding this. do you have a discord or something? email?

Sorry! not an issue. Sincerely apologize for creating one but there's not many people with the know how of mipi who are able to talk about it so i'd greatly appreciate it once again if you could help me out a little here. I'd love to contribute something to any of your projects in exchange. :)

twlostow, 2021-02-04:
What's Discord?

HugoGresse, 2022-01-05:
Discord is a messaging service like Slack, Messenger, etc 
https://discord.com/